PolarMesosphericClouds/PMC-Turbo #27: missing disk on pmc-camera-2

gitj, 2017-08-01:
pmc-camera-2 has these disks
<code>
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0USS7Y -> ../../sdc
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0UWTEY -> ../../sdd
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0U206Y -> ../../sda
</code>
but fstab shows these entries:
<code>
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0UEKSY-part3 /data1 ext4 nofail 0 2
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0U206Y-part3 /data2 ext4 nofail 0 2
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0USS7Y-part1 /data3 ext4 nofail 0 2
/dev/disk/by-id/ata-WDC_WD80EFZX-68UW8N0_VK0UWTEY-part1 /data4 ext4 nofail 0 2
</code>

The data1 disk VK0UEKSY seems to be missing.

gitj, 2017-08-30:
Appears that SATA cable was strung too tightly and loosened from motherboard connection

gitj, 2017-09-14:
/data1 (serial number VK0UEKSY) is missing again. Need to check if cable needs to be replaced.

ckjellstrand, 2017-09-14:
The cable is completely unplugged.

ckjellstrand, 2017-09-26:
Cable was plugged in. Since then the OS has been completely re-installed and disks reformatted multiple times.

closed 2017-09-26 by ckjellstrand