PolarMesosphericClouds/PMC-Turbo #41: finish lowrate monitor gui improvements

gitj, 2017-09-15:
add colors and value formatting for new table based gui to make it as informative as original

ckjellstrand, 2017-09-15:
- Make short_status_format or short_status_ranges.py which contains OrderedDict of (item name from short status, human readable item name, formatting (e.g. %.2f), good range, bad range)
 - Write a function to take in an item name from short status, and output the human readable name, and a value which is correctly formatted and ready for input into the table of lowrate_monitor_gui
 - Make lowrate_monitor_gui highlight columns (background color) which are new, old, etc.

ckjellstrand, 2017-09-15:
Formatting needs to be more than just string formatting - needs to be a function (for example, to divide value by 1000 to get easily readable units)

ckjellstrand, 2017-09-18:
Most of this is closed. Need to find good names and formatting for several items.

closed 2017-09-20 by ckjellstrand