PolarMesosphericClouds/PMC-Turbo #53: Change focus step in pgcamera to be more readable

ckjellstrand, 2017-09-19:
Right now 0 -> 0 focus step, 1000 -> max focus step. This isn't useful for sending change focus commands. Look for a place where this information is sent to ground or allow it to be entered manually upon starting pgcamera.

ckjellstrand, 2017-09-22:
Approximate focus step in there. Easy enough to make a command line argument but that seems annoying each time to start up. pgcamera looks at JPEGFiles while the max focus step is in ShortStatus messages, and I don't want pgcamera to be looking at ShortStatus messages unless we add a lot of features.

gitj, 2017-09-22:
Does this work for wide and narrow field cameras?

ckjellstrand, 2017-09-26:
Added for narrow field camera as well

ckjellstrand, 2017-09-26:
Closed by 12cb1381af0369367a32bf77e8b1a6ad9755f742

closed 2017-09-26 by ckjellstrand